espressif/idf-im-ui #71: Ubuntu 20.04 Support?

mmrein, 2025-04-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Just tried the tool on Ubuntu 20.04 (which is now almost "EOL") and it won't run, both GUI nad CLI.

**Trying to run  GUI from terminal I get:**

```
./eim: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory
```

**And running CLI from terminal gives:**

```
./eim: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by ./eim)
./eim: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./eim)
./eim: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ./eim)
./eim: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.39' not found (required by ./eim)
./eim: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./eim)
./eim: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./eim)
```

Not a big deal, I don't really need the tool on its own and probably going to update the Ubuntu VM at some point, just wanted to let you know.

Hahihula, 2025-04-08:
We know about this issue. It's caused by dynamically linked glibc and further dependencies. Linking the glibc statically is on top of our backlog, just after some freshly found bugs. Thanks for letting us know, and I will post an update here after this issue is resolved.

mmrein, 2025-04-17:
@Hahihula  I'll probably wait for the release, but thanks for letting me know!

Hahihula, 2025-04-22:
> [@Hahihula](https://github.com/Hahihula) I'll probably wait for the release, but thanks for letting me know!

Hi @mmrein ,
The release was done today -> https://github.com/espressif/idf-im-ui/releases/tag/v0.2.1 
Alternatively you can download from our [download page](https://dl.espressif.com/dl/eim/)

closed 2025-04-22 by Hahihula